SensorsIot/DCF77-Transmitter-for-ESP32 #7: Missing cronCheck() function.

wip58, 2022-12-07:
Hi,
thank you for this excellent program! Works really nice.

I would like to use the CONTINUOUSMODE option, but there is a reference to cronCheck() function, which is missing in the code.

Do you have it in your notes, somewhere? I am not capable to write it myself, yet :)

Best Regards,
Wieslaw

Mercoory, 2022-12-07:
The referenced cronCheck() function is in the SLEEP_CRON.ino file.

wip58, 2022-12-07:
O! Thank you very much for pointing me into the right direction!

deltazerorsan, 2022-12-07:
> Hi, thank you for this excellent program! Works really nice.
> 
> I would like to use the CONTINUOUSMODE option, but there is a reference to cronCheck() function, which is missing in the code.
> 
> Do you have it in your notes, somewhere? I am not capable to write it myself, yet :)
> 
> Best Regards, Wieslaw

Hi, 

To use CONTINUOUSMODE you only need to uncomment line 29,

`// #define CONTINUOUSMODE // Uncomment this line to bypass de cron and have the transmitter on all the time`

That's all!

Cheers

wip58, 2022-12-07:
Thanks @deltazerorsan,
@Mercoory pointed me to the right direction as the cronCheck() was in another file SLEEP_CRON.ino.
I still have to work around that on the "hour"part as only "minutes" puts the ESP into sleep mode.
Thank you for reaching out :)

Best,

wip58, 2022-12-08:
My goal is to have the unit working for about 20 minutes after power on. Then it should go to sleep and wake up at 1, 2, 3 am (5min before till 8 min past the hour as in the original code). This is when my watch syncs automatically.
It seems that the original code already does this. I need to adjust the timers and hours to my needs.
I added an OLED display to see what is going on, displaying the time, date and a progress bar. My watch seems to take 3 "packets" of the DCF signal, assuming they are correct and only then adjust the time, hence the bar for 3 minutes. There is also an offset in the box, just to play with the signal and see watches' reaction (it works). I can choose -3h, +2h, or 0 = accurate for my timezone, by pressing a pushbutton. (picture)
[image]

It seems that now everything is working, however there is a 20 second difference between the OLED clock and the watch time after the unit wakes from sleep. I will investigate more and maybe call NTP again after each wakeup...

deltazerorsan, 2022-12-08:
Hi @wip58 ,

`34 #define minuteToSleep   8 // If it is running at this minute then goes to sleep and waits until minuteToWakeUp`

Here you can change the minutr the ESP32 goes to sleep; easy change.

The code goes to catch NTP every run, see line 73 inside the setup, then shuts down the WiFi and print the time to the monitor so you can check if that time is correct or not.

I had observed a similar delay when an ESP32 wakes up, but never if you press reset. Maybe you are in thast case.

About timezone you can choose the correct one for you including DST dates in line 53; you don't need to add nothing more to have the correct timezone.

Good luck

deltazerorsan, 2022-12-08:
Hi (again) @wip58 ,

What kind of watch are you using?

deltazerorsan, 2022-12-08:
> Thanks @deltazerorsan, @Mercoory pointed me to the right direction as the cronCheck() was in another file SLEEP_CRON.ino. I still have to work around that on the "hour"part as only "minutes" puts the ESP into sleep mode. Thank you for reaching out :)
> 
> Best,

Ouch! Have not reading this.

You don't need the hour because if the ESP32 is awake the minute (8 inthe original code) puts the system to sleep.

In the wrong hour the system is not awake so nothing to do...

wip58, 2022-12-08:
Hi @deltazerorsan,

_I had observed a similar delay when an ESP32 wakes up for it self but never if you press reset. Maybe you are in thast case._ Indeed, my case.

_About timezone you can choose the correct one for you including DST dates in line 53; you don't need to add nothing more to have the correct timezone._ Already did. The offset is just for fun ;)

_In the wrong hour the system is not awake so nothing to do..._ Just checks cron and goes to sleep again.

Thank you for your help.
Best

I use this one: 
[image]

closed 2024-11-20 by SensorsIot