SensorsIot/DCF77-Transmitter-for-ESP32 #8: No sync but high RC level

m-kalendarev, 2023-01-16:
Hi. Tried to get worked on ESP-WROOM-32 but got sync error. I have several Casio RC watches, I can get time with the real Germany signal but got an error with emulated. The watches shows high receiving level, I can see generated time messages in console, even LED is blinking when added to the antenna wire. The antenna is a 20 wire turns around the watches, just for the test. 
Wifi is connected, ntp time is ok.

deltazerorsan, 2023-01-16:
Hi @m-kalendarev,

Have you tried with less signal? Maybe there is so much signal due to have the watches just in the antenna...

Is the receiving level following the pulses?

m-kalendarev, 2023-01-16:
Yes I tried, but no luck. I've watched #287 video to understand the antenna connection but the video is about raspberry pi (2 different GPIO outputs connected to antenna), the esp32 has only 1 output to antenna, and I connect it to the coil with 400 ohm   resistor, the other side to the gnd.

SensorsIot, 2023-01-16:
If the Seikos work, I think, your transmitter works, too. Where else should these watches get the signal?
So the problem is between the transmitter and the Casios or at the Casios. If you have an oscilloscope, I would try to measure the frequency of the transmitter. Maybe it is wrong and the Casios have a better filter. You should also see how the signal looks like.

deltazerorsan, 2023-01-17:
> Yes I tried, but no luck. I've watched #287 video to understand the antenna connection but the video is about raspberry pi (2 different GPIO outputs connected to antenna), the esp32 has only 1 output to antenna, and I connect it to the coil with 400 ohm resistor, the other side to the gnd. upd: Tried all my watches, 2 Seikos and 3 Casios. Seiko gets update quickly, 2-3 min, but Casio shows high reception level for 5-7 min and return err. I have no idea what is the difference because all this watches successfully updated 3-4 times a week with native DE signal placed near the window at night.

There is a big difference between the raspi and ESP32 implementations. My two Junghans syncronize without trouble with the raspi one but not with the ESP32 one.

In the raspi there is a full signal when pulsing and a 20% (aprox) signal when no pulse. In the ESP32 there is full signal when pulse and NO signal at all when no pulse. That simpler implementation IS NOT the standard and some watches lost syncro because it depends on the carrier frecuency that disapear during no pulse in the case of the ESP32 implementation.

I have a Brigmton Weather Station and some chinesse movements that works with both implemantation, so there is for sure some, as Andreas said, better filtering or more "on the standard" implementations in those watches.

I use the same antenna for both but it is more sofisticated: ferrite, two layers of turns and a capacitor in parallel...

Some time ago I was thinking to adapt the ESP32 implo to the standar but some lack of time and interest...

m-kalendarev, 2023-01-17:
Hi, thank you for the clarification, I was thinking about it yesterday). Just for an experiment will try raspi with my Casio watches. And will order ferrite rod for LC antenna. The goal was to built in little low consumption transmitter in the wooden watch box and synch all my watches.

deltazerorsan, 2023-01-17:
Sure mine is not miniaturized...

[image]

;-)

SensorsIot, 2024-03-12:
Thank you for your write-up. It could help somebody with the same problem. Because this is an old project (and it still works here), I will most probably not return to it.

closed 2024-03-12 by SensorsIot