SensorsIot/DCF77-Transmitter-for-ESP32 #9: The antenna and two GPIOs

jm314159, 2024-07-21:
Hi, Andreas,
following the question in #5, please, once again, share the idea of how have you connected the antenna circuit shown in video 287 to ESP32. In the video, which is for Raspberry, you are using **two** GPIOs: GPIO 4 and GPIO 17, while in the ESP32 code you only have **one** `#define ANTENNAPIN 15`.

Referencing to video 287, I would say that ANTENNAPIN should be related to GPIO 17. I am not sure if I correctly understand the statement: "GPIO 4 is used to create a fixed frequency". How is that achievied? Just by setting the state to high? Or am I wrong? Please comment on that.

Btw., since esp32 v3.0.0, these functions `ledcSetup` and `ledcAttachPin` have been removed and substituted with `ledcAttach` that merges the two former functions. To compile the code, I commented out the corresponding two lines:
```
//  ledcSetup(0, 77500, 8); // DCF77 frequency
//  ledcAttachPin(ANTENNAPIN, 0); // This Pin, or another one you choose, has to be attached to the antenna
```
and used the following one instead:
`ledcAttach(ANTENNAPIN, 77500, 8);`

jm314159, 2024-07-23:
Nvm., I managed to run the circuit. Most important items for me in the connection and setup area are the following:

1. Antenna
1.1. Connection circuit based on [Video 287, time 9'50''](https://youtu.be/6SHGAEhnsYk?t=590), "Remote Controller for Clocks"
1.2. My antenna inductance 4.60 mH, measured as in [Video 287, time 6'40''](https://youtu.be/6SHGAEhnsYk?t=400). Capacitor 0.93 nF (measured, ref. 1 nF)
2. ESP32 board by Espressif
2.1. No onboard LED, I used the external one - connected to GPIO 5
2.2. Antenna attenuation (background) pin - GPIO 4
2.3. Antenna pulse (main) pin - GPIO 17
3. In Arduino code: "IDE, DFC77_ESP32.ino"
3.1. Definitions in a global section
```
#define LEDBUILTIN 5      // External LED
#define ANTENNAATT 4      // Antenna attenuation
#define ANTENNAPIN 17     // Antenna frequency output
```
3.2. In `void setup()`
```
  pinMode (LEDBUILTIN, OUTPUT);
  digitalWrite (LEDBUILTIN, HIGH);  // HIGH, inversely, switches the LED off

  pinMode(ANTENNAATT, OUTPUT);
  digitalWrite (ANTENNAATT, HIGH);
```
The original [hzeller's project](https://github.com/hzeller/txtempus/blob/master/img/schematic-dcf77.png) is for Raspberry Pi, where GPIOs up to 8 have the default state HIGH. So, GPIO 4, pin 7 on RPi, referenced in [Video 287, time 8'45''](https://youtu.be/6SHGAEhnsYk?t=525), has 3.3 V. That's why, I have set ANTENNAATT HIGH for ESP32.

3.3. In `void DcfOut()`, I have toggled all `digitalWrite(LEDBUILTIN, HIGH);` to `digitalWrite(LEDBUILTIN, LOW;`, and vice-versa.

4. In Arduino Board Manager, I installed esp32 by Espressif in version 2.0.17. Unfortunately, since v3.0.0, `ledcSetup` and `ledcAttachPin` were removed. A simple redefinition as I mentioned above was insufficient and the circuit did not work.

As a final remark, I can say that I received the simulated DCF signal to update clocks ca. 4 m away from my coil.

SensorsIot, 2024-07-23:
I'm sorry for not getting back to you sooner. I was on a trip. I hope your setup works now.
The Pi produces the RF signal differently. One pin is always on, and a second (the attenuator pin) reduces the signal according to the DCF77 signal.
The ESP32 switches the RF signal according to the DCF77 signal on and off. It, therefore, does not need the attenuator pin or the 50 ohms resistor.

jm314159, 2024-07-23:
Confirm, by experiment, the circuit also works fine without 50 ohm and with GPIO 4 disconnected (neglected).
On the other hand, when I connected 5 V to Vcc, then one clock has got updated at the distance of about 9 m as well as another one located 3 m beneath - through the ceiling.

SensorsIot, 2024-07-23:
Cool! The output power is increased if you connect the transistor to 5V. This is the reason for my remark on YouTube.

closed 2024-07-23 by SensorsIot