SensorsIot/DCF77-Transmitter-for-ESP32 #11: Power saving proposal: reduce cpu frequency

IU5HKU, 2024-11-19:
This isn't an issue, your code works great, my old oregon scientific clock is rebirth :-)
I'm using this code with a battery operated esp32 who sits near the clock, with 4 NiMh batteries (i know, it's not the best, but it's a zero-cost project and i've used things that i had in my junkyard, i'm using an old ferrite antenna from another dead clock...). To maximize batteries life i've reduced the cpu frequency with this code into setup():

  if(setCpuFrequencyMhz(80)){
    Serial.print("CPU frequency set @");
    Serial.print(getCpuFrequencyMhz());
    Serial.println("Mhz");
  }
  else
    Serial.println("Fail to set cpu frequency");

...why not add this to your code... :-)
Have a nice day
 MC

deltazerorsan, 2024-11-19:
> This isn't an issue, your code works great, my old oregon scientific clock is rebirth :-) I'm using this code with a battery operated esp32 who sits near the clock, with 4 NiMh batteries (i know, [...]

Hi,

Are you using all the features about timing the periods of transmission?

There is a few nice features to save power...

deltazerorsan, 2024-11-20:
Nice to hear that!

closed 2024-11-20 by SensorsIot