SensorsIot/DCF77-Transmitter-for-ESP32 #12: Casio G-Shock and ESP32 C3

cino893, 2025-02-09:
Do somebody knows ot have some ideas why this code is running on ESP32 and after changing port to one I've added some cable it seen as very strong on my Casio G-Shock but on the other hand it's not able to set date?

Mercoory, 2025-02-09:
Check issue [#8](https://github.com/SensorsIot/DCF77-Transmitter-for-ESP32/issues/8)

SensorsIot, 2025-02-09:
Strange. You should see the signal output in the Serial monitor. So you can check if the signal is correct.
I never tested it on a -C3. So, if you have an ESP32 around, I would try this one with the original sketch.

cino893, 2025-02-09:
Thank you for response.
I sounds that its all about #8, do you know how to limit the power to 20% instead of shuting it down?
PWM even on 20% usage will generate 100% amplitude propably.

SensorsIot, 2025-02-09:
I would extend the distance if you fear a too-strong signal. You also can add a resistor in series that will burn some of the power.

cino893, 2025-02-10:
It's not about strength of signal. It's about being compliant with the DCF77 standard, a d amplitudę modulation for 1 it's not turned off but amplitude is generated down to 15%

[image]

cino893, 2025-02-10:
I finally did it! 🥳

It's a little dirty(so I'm not making pull request) in repo but it's working I did by constatnly sending from PWM 77.5kHz and manipulating ground by DRAIN pins

One drain is full power and the send one 10% divider

You can check it here:
https://github.com/cino893/DCF77-Transmitter-for--XIAO-ESP32-C3

closed 2025-12-05 by SensorsIot