SensorsIot/DCF77-Transmitter-for-ESP32 #13: Not syncing correctly when waking up from deep sleep

Astrall2, 2025-05-22:
Thank you for this great project!

During testing, I found that when using the ESP32 to wake up at specific hours, the NTP sync does not work correctly. Instead, it retrieves the time from the ESP32’s internal RTC. In my case, this results in a drift of approximately 1 minute after 1 hour of sleep.

You can reproduce this behavior by enabling timestamp logging in Arduino.

As a workaround, I managed to resolve the issue by manually setting the RTC to an invalid value, which forces a proper NTP sync on wake-up.

I added the following code to the setup() function:

```
struct timeval tv = { .tv_sec = -1, .tv_usec = 0 };
settimeofday(&tv, nullptr);
```

If you have a more elegant solution, I would be happy to hear it.

Kind Regards
George

gpregger, 2025-06-29:
Hey, I'm having the same issue and tried your workaround. Unfortunately with your code added the time started to be all over the place for me.  
After a bit of research I've now tried instead increasing the wait time inside the getNTP function, which now looks like this:

```c
void getNTP() {
    Serial.print("GetNTP ");
    int i = 0;
    do {
        i++;
        if (i > 40) {
            ESP.restart();
        }
        setenv("TZ", TZ_INFO, 1);
        configTzTime(TZ_INFO, ntpServer);
        delay(5000);
    } while (!getLocalTime(&timeinfo));
    Serial.println("Ok");
}
```
(I've also replaced configTime with configTzTime, but I don't know if that made any difference)

So far my ESP has been running for about 12h (20min per hour, the rest in deep sleep) and the time is still spot on.  
If I understand correctly the NTP sync runs in the background asynchronously, so this is probably still not a bulletproof solution.

If my current attempt doesn't end up working well, I'll probably try involving `esp_sntp.h` which may allow forcing an immediate sync by doing `sntp_set_sync_mode(SNTP_SYNC_MODE_IMMED)` and "catch" a successful sync by attaching a callback via `sntp_set_time_sync_notification_cb`. But I haven't fruther explored that at all yet.

Note that I'm running this on an M5Stack Atom from my fork: https://github.com/gpregger/DCF77-Transmitter-for-ESP32

SensorsIot, 2025-12-05:
Thanks for the workaround! Documented in the README under Troubleshooting.

See: https://github.com/SensorsIot/DCF77-Transmitter-for-ESP32#deep-sleep-ntp-sync-issue